tube0013/tube_gateways #209: how to change the zwave tcp port in the zwave poe kit?

ltgcc, 2024-11-01:
I have the zwave poe housing with the zooz zac93 GPIO in it.   How do I change the port for the zwave from 6638/tcp to <something else>/tcp ?   I assume this is part of the ESP functionality.  And I've found

https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-zw-kit/tubeszb-zw.yaml

I assume I need to change:

stream_server:
  buffer_size: 2048
  uart_id: uart_bus
#  id: ss
  port: 6638 # optional, default is 6638   **<- this line.**

but maybe not?

I apologize up front, this seems like one of those things that should be obvious.  The page says instructions on how to change  instructions are under the individual product pages.  I'm not seeing it or finding how to do it there or anywhere.  

It doesn't appear to be on the zwave POE webpage, and there's no telnet or ssh.  I'm at a loss.  Can you point me in the right direction?

Thanks much in advance

tube0013, 2024-11-01:
You would need to set up ESPHome standalone or addon-on if using HAOS. Grab the yaml file from here and put it in your ESPHome folder, change the port in the line you found, save and compile and install the new firmware.

ltgcc, 2025-02-26:
followup, very late, apologies.   Turns out I figured out how to -not- need to change the port number.  I am still using :6638.  Thanks!

closed 2025-02-26 by ltgcc