tube0013/tube_gateways #212: Unable To Update Firmware

D3v01dZA, 2024-11-06:
To start I'm not some ESP wizard and have no idea of how it works.

I have this adapter https://tubeszb.com/product/efr32-mgm24-poe-coordinator/. I am currently unable to update it and I cannot work out why, I am using the firmwares from https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp. Specifically I wanted to try ```tubesZB-EFR32-MGM24_NCP_7.4.3.gbl```.

I've tried a couple different methods from https://www.youtube.com/watch?v=zKrISuWEzL4 to https://github.com/NabuCasa/universal-silabs-flasher/tree/dev/universal_silabs_flasher - I run a docker based HASS so the add on is a bit painful but looking at the code it just delegates to this.

When I tried https://www.youtube.com/watch?v=zKrISuWEzL4 everything "looks" normal and I see the little ~ and everything, then when I try to start Zigbee2MQTT it still fails with ```2024-11-06 16:13:30] error: 	z2m: Error: Adapter EZSP protocol version (11) is not supported by Host [13-14].```. It's very unclear where anything is failing or how the firmware is actually written, and there are no errors/output during any steps.

When trying universal_silabs_flasher the same story mostly, no errors, sometimes flash uploads and does nothing, sometimes it spews out "No Firmware"

Probe output - this is the same output no matter what I do, post firmware "upgrade" etc.
```
└─(15:58:47)──> universal-silabs-flasher --device socket://zigbee.lan:6638 probe                                                                                                                                                                                                                                                                                                 
2024-11-06 15:58:54.030 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-06 15:58:56.049 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 460800 baud
2024-11-06 15:59:00.381 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 115200 baud
2024-11-06 15:59:04.704 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.CPC at 230400 baud
2024-11-06 15:59:09.083 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2024-11-06 15:34:42.372 Chris-MacBook.local universal_silabs_flasher.flasher INFO Detected ApplicationType.EZSP, version '7.2.3.0 build 215' (7.2.3.0.215) at 115200 baudrate (bootloader baudrate None)
2024-11-06 15:34:42.372 Chris-MacBook.local universal_silabs_flasher.flash INFO Dumping EmberZNet Config
CONFIG_PACKET_BUFFER_COUNT=255
CONFIG_NEIGHBOR_TABLE_SIZE=26
...
```

Playhing with the bootloader pin I sometimes get
```
└─(16:04:35)──> universal-silabs-flasher --probe-method bootloader --device socket://zigbee.lan:6638 flash --firmware tubesZB-EFR32-MGM24_NCP_7.4.3.gbl                                                                                                                                                                                                                           1 ↵ ──(Wed,Nov06)─┘
2024-11-06 16:04:39.993 Chris-MacBook.local universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.3', ezsp_version='7.4.2.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2024-11-06 16:04:39.993 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-06 16:04:41.050 Chris-MacBook.local universal_silabs_flasher.flasher INFO Detected bootloader version '2.1.1'
2024-11-06 16:04:41.051 Chris-MacBook.local universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '2.1.1' at 115200 baudrate (bootloader baudrate 115200)
tubesZB-EFR32-MGM24_NCP_7.4.3.gbl  [####################################]  100%
Traceback (most recent call last):
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/universal_silabs_flasher/flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/asyncio/base_events.py", line 685, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/universal_silabs_flasher/flash.py", line 423, in flash
    await flasher.flash_firmware(
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/universal_silabs_flasher/flasher.py", line 299, in flash_firmware
    await gecko.run_firmware()
  File "/Users/christopheraltona/.asdf/installs/python/3.12.2/lib/python3.12/site-packages/universal_silabs_flasher/gecko_bootloader.py", line 101, in run_firmware
    raise NoFirmwareError("No firmware exists on the device")
universal_silabs_flasher.gecko_bootloader.NoFirmwareError: No firmware exists on the device
```

But mostly I get something that looks "successful" but the probe output and zigbee2MQTT error remains the same.

```
└─(15:00:54)──> universal-silabs-flasher --device socket://zigbee.lan:6638 flash --firmware tubesZB-EFR32-MGM24_NCP_7.4.3.gbl      
2024-11-06 15:01:13.851 Chris-MacBook.local universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.3', ezsp_version='7.4.2.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant=None, baudrate=115200)
2024-11-06 15:01:13.851 Chris-MacBook.local universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2024-11-06 15:01:13.907 Chris-MacBook.local universal_silabs_flasher.flasher INFO Detected bootloader version '2.1.1'
2024-11-06 15:01:13.907 Chris-MacBook.local universal_silabs_flasher.flasher INFO Detected ApplicationType.GECKO_BOOTLOADER, version '2.1.1' at 115200 baudrate (bootloader baudrate 115200)
tubesZB-EFR32-MGM24_NCP_7.4.3.gbl  [####################################]  100%
```

D3v01dZA, 2024-11-06:
Bellows thinks I'm still on 7.2. I am actually trying to migrate from ZHA to Z2MQTT. I had both containers down while I was working on this, could confirm on the sticks webpage where it showed 0 active connections and serial not connected.

After the upload using Serial it took a little bit to show the ~ but it was quicker than the upload. The universal-silabs-flasher just returns quickly after the upload.

Using Serial I followed your YouTube exactly (although that was a different controller). Am I supposed to pull the boot loader pin for this flasher to work? I'm a bit stumped :)

```
└─(17:47:48)──> bellows --baudrate 115200 -d socket://zigbee.lan:6638 info 
[00:21:2e:ff:ff:09:4b:93]
[0x0000]
[<EmberNetworkStatus.JOINED_NETWORK: 2>]
[<EmberStatus.SUCCESS: 0>, <EmberNodeType.COORDINATOR: 1>, EmberNetworkParameters(extendedPanId=74:94:a3:5c:14:9c:f1:2b, panId=0xE7D7, radioTxPower=8, radioChannel=20, joinMethod=<EmberJoinMethod.USE_MAC_ASSOCIATION: 0>, nwkManagerId=0x0000, nwkUpdateId=0, channels=<Channels.ALL_CHANNELS: 134215680>)]
[<EmberStatus.SUCCESS: 0>, EmberCurrentSecurityState(bitmask=<EmberCurrentSecurityBitmask.GLOBAL_LINK_KEY|HAVE_TRUST_CENTER_LINK_KEY|TRUST_CENTER_USES_HASHED_LINK_KEY|96: 244>, trustCenterLongAddress=00:21:2e:ff:ff:09:4b:93)]
Manufacturer: TubesZB
Board name: Series 2
EmberZNet version: 7.2.3.0 build 215
```

tube0013, 2024-11-06:
as long as there is a working ezsp firmware on the device, the universal-silabs-flasher should be able to put the device into bootloader mode with ezsp commands with you not having to do anything else. 

I have been running and now recommend this FW:
https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp/4.4.4

can you try and run:

`universal-silabs-flasher -vv --device socket://zigbee.lan:6638 --bootloader-baudrate 115200 --ezsp-baudrate 115200 flash --firmware ./tubeszb-mgm24-hw-max_ncp-uart-hw_7.4.4.0.gbl --force`

D3v01dZA, 2024-11-06:
I would prefer not to brick my device, can --force do something weird? (Speaking from a little bit of experience with --force equivalents that made me sad :()

tube0013, 2024-11-06:
Then just leave that off. I use it so I can flash different builds of the same fw version

D3v01dZA, 2024-11-06:
Didn't see to take again. Killed my ZHA - checked 0 connections on the web page as well as no serial user. I waited a good 5 minutes after typing this original draft then ran bellows again.
```
universal-silabs-flasher -vv --device socket://zigbee.lan:6638 --bootloader-baudrate 115200 --ezsp-baudrate 115200 flash --firmware ./tubeszb-mgm24-hw-max_ncp-uart-hw_7.4.4.0.gbl
```
[Attached Logs](https://github.com/user-attachments/files/17654211/upload.log)
```
bellows --baudrate 115200 -d socket://zigbee.lan:6638 info
```

tube0013, 2024-11-07:
are you 100% confident you have an mgm24? the board name shown is not what I program on the mgm24 modules, that is what I put on the mgm21 modules.

can you take a picture of the module so I can confirm? or do you see a ceramic antenna on the the board? if so it's a mgm21. and it is likely rejecting the fw for that reason.

D3v01dZA, 2024-11-07:
>I'm pretty sure whatever the mistake is I'm making is simple and obvious

Remember when I said this? That is the mistake. I am not exactly sure _how_ I thought I had the MGM24 but a search in my inbox initially showed that - I rechecked now and it is indeed an MGM21 :)

The flash went smoothly once I downloaded the right firmware!

closed 2024-11-07 by D3v01dZA