tube0013/tube_gateways #217: CC2652P2 based Zigbee to Ethernet/USB Serial Coordinator stays off, zigbee2mqtt can't connect to the adapter

shad-lp, 2025-01-04:
Hello,

I'm facing issues with this adapter I bought in 2022, Zigbee2MQTT says can't connect to the adapter.
I tried to restart it from its webserver (it was ON), now it stays OFF and restarting it doesn't change anything.
I also tried to reset the Zigbee module, no effect.
I finally unplugged it (USB), no more luck.

Any advice ? Is it dead ? or maybe a firmware upgrade could solve the case ?

tube0013, 2025-01-04:
What are the z2m logs showing?

shad-lp, 2025-01-04:
Here there are :

```
2025-01-04T13:36:09.846764625Z Using '/app/data' as data directory
2025-01-04T13:36:10.216783635Z Starting Zigbee2MQTT without watchdog.
2025-01-04T13:36:18.957082498Z [2025-01-04 14:36:18] debug: 	z2m: Removing old log directory '/app/data/log/2025-01-04.14-34-21'
2025-01-04T13:36:18.970635493Z [2025-01-04 14:36:18] info: 	z2m: Logging to console, file (filename: log.log)
2025-01-04T13:36:18.987959731Z [2025-01-04 14:36:18] debug: 	z2m: Loaded state from file /app/data/state.json
2025-01-04T13:36:19.066795030Z [2025-01-04 14:36:19] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
2025-01-04T13:36:19.069227094Z [2025-01-04 14:36:19] debug: 	z2m: sd-notify loaded
2025-01-04T13:36:19.079171235Z [2025-01-04 14:36:19] info: 	z2m: Starting zigbee-herdsman (3.2.1)
2025-01-04T13:36:19.085178208Z [2025-01-04 14:36:19] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"tcp://192.168.150.70:6638\"},\"adapter\":{\"disableLED\":false}}"'
2025-01-04T13:36:19.100888398Z [2025-01-04 14:36:19] error: 	z2m: Error while starting zigbee-herdsman
2025-01-04T13:36:19.109362564Z [2025-01-04 14:36:19] error: 	z2m: Failed to start zigbee-herdsman
2025-01-04T13:36:19.114739318Z [2025-01-04 14:36:19] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
2025-01-04T13:36:19.116272311Z [2025-01-04 14:36:19] error: 	z2m: Exiting...
2025-01-04T13:36:19.404155704Z [2025-01-04 14:36:19] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
2025-01-04T13:36:19.404296777Z     at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
2025-01-04T13:36:19.404419499Z     at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
2025-01-04T13:36:19.404466240Z     at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
2025-01-04T13:36:19.404551332Z     at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
2025-01-04T13:36:19.404607313Z     at Zigbee.start (/app/lib/zigbee.ts:69:47)
2025-01-04T13:36:19.404646776Z     at Controller.start (/app/lib/controller.ts:142:13)
2025-01-04T13:36:19.404685091Z     at start (/app/index.js:161:5)
```

shad-lp, 2025-01-04:
And the TB coordinator webpage :

[image]

shad-lp, 2025-01-04:
I stumbled upon this : https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#cc2652-firmware-flashing

Could it be useful ? I flashed it this way once a year and a half maybe, that's what you told me to do. Had almost the same problem, but can't remember exactly.

katekyonni, 2025-01-04:
Z2M 2.0 has some breaking changes, that I would recommend reading. https://github.com/Koenkk/zigbee2mqtt/discussions/24198

You need to specify the adapter type in your Z2M configuration, since `zstack` is no longer the default. (https://github.com/Koenkk/zigbee2mqtt/discussions/24364)

Example:
```
serial:
  port: tcp://192.168.150.70:6638
  adapter: zstack
  ```

shad-lp, 2025-01-04:
@katekyonni Thx a lot, that was the root of my issue.
Watchtower ran its update process this morning and basically upgraded to 2.0.0
I made the changed you suggested and it works.
I'll give a deeper look at the changes notes.
Sorry for the loss of time and thank you guys !

closed 2025-01-04 by shad-lp