tube0013/tube_gateways #219: Z-Wave PoE Kit - Errors in log after flashing static IP

tfberg, 2025-01-08:
I flashed the device this morning via USB using the firmware.factory.bin file generated via ESPHome Builder. I watched the log for a while to see if any issues popped up. Saw none. Just looked again and I am seeing these. Please advise.

[image]

tube0013, 2025-01-08:
I've not see these before. They are related to the frontend webserver on the device and should not affect the overall performance of the device. as long as you are not connected to the web frontend which is really only needed when troubleshooting, you should be fine.

tfberg, 2025-01-08:
Okay, thanks.