tube0013/tube_gateways #220: Update tubeszb-mgm24 firmware to 4.4.4

jkosharek, 2025-01-21:
I am using the TubesZB Silicon Labs FW Flasher add-on in HAOS, I put in my network device, 10.11.13.196:6638, selected '/dev/ttyS1' and put in the firmware url:
https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp/4.4.4/tubeszb-mgm24-hw-max_ncp-uart-hw_7.4.4.0.gbl

Logs:
```
-----------------------------------------------------------
 Add-on: TubesZB Silicon Labs FW Flasher
 TubesZB Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.4.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2025.1.3
 Home Assistant Supervisor: 2024.12.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[13:30:23] INFO: Starting universal-silabs-flasher with socket://10.11.13.196:6638 (bootloader baudrate 115200)
Error: '/root/firmware.gbl' does not appear to be a valid firmware image: ValueError('Unknown firmware image type')
[13:30:24] INFO: universal-silabs-flasher-up script exited with code 1
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 1
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service banner: starting
```

tube0013, 2025-01-21:
you must copy and use the Raw Link. I'll try and update the addon to pull a latest image down in the future, but I have to worry about different models too.

https://github.com/tube0013/tube_gateways/raw/refs/heads/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp/4.4.4/tubeszb-mgm24-hw-max_ncp-uart-hw_7.4.4.0.gbl

jkosharek, 2025-01-21:
Changed the URL to:
https://github.com/tube0013/tube_gateways/raw/refs/heads/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp/4.4.4/tubeszb-mgm24-hw-max_ncp-uart-hw_7.4.4.0.gbl

and now I am getting this:
```
-----------------------------------------------------------
 Add-on: TubesZB Silicon Labs FW Flasher
 TubesZB Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.4.0
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.1  (amd64 / generic-x86-64)
 Home Assistant Core: 2025.1.3
 Home Assistant Supervisor: 2024.12.3
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[13:54:38] INFO: Starting universal-silabs-flasher with socket://10.11.13.196:6638 (bootloader baudrate 115200)
2025-01-21 13:54:39.326 d0f5a498-tzb-silabs-flasher asyncio DEBUG Using selector: EpollSelector
2025-01-21 13:54:39.326 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.4', ezsp_version='7.4.4.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.NCP_UART_HW: 'ncp-uart-hw'>, baudrate=115200)
2025-01-21 13:54:39.327 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flash DEBUG Probing app type ApplicationType.EZSP first
2025-01-21 13:54:39.327 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flash DEBUG Probing with 115200 baudrate first
2025-01-21 13:54:39.327 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-01-21 13:54:39.327 d0f5a498-tzb-silabs-flasher zigpy.serial DEBUG Opening a serial connection to 'socket://10.11.13.196:6638' (baudrate=115200, xonxoff=False, rtscts=False)
2025-01-21 13:54:39.328 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.common DEBUG Connection made: <_SelectorSocketTransport fd=3 read=idle write=<idle, bufsize=0>>
2025-01-21 13:54:39.328 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.common DEBUG Sending data b'\n'
2025-01-21 13:54:39.328 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.common DEBUG Sending data b'3'
2025-01-21 13:54:41.331 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
2025-01-21 13:54:41.332 d0f5a498-tzb-silabs-flasher zigpy.serial DEBUG Opening a serial connection to 'socket://10.11.13.196:6638' (baudrate=115200, xonxoff=True, rtscts=False)
2025-01-21 13:54:41.347 d0f5a498-tzb-silabs-flasher bellows.uart DEBUG Connection lost: ConnectionResetError('Remote server closed connection')
2025-01-21 13:54:41.347 d0f5a498-tzb-silabs-flasher bellows.uart ERROR Lost serial connection: ConnectionResetError('Remote server closed connection')
2025-01-21 13:54:41.348 d0f5a498-tzb-silabs-flasher bellows.ezsp DEBUG socket://10.11.13.196:6638 connection lost unexpectedly: Remote server closed connection
2025-01-21 13:54:41.348 d0f5a498-tzb-silabs-flasher bellows.ezsp INFO NCP entered failed state. No application handler registered, ignoring...
2025-01-21 13:54:41.348 d0f5a498-tzb-silabs-flasher bellows.uart DEBUG Connection lost: None
2025-01-21 13:54:41.348 d0f5a498-tzb-silabs-flasher bellows.uart DEBUG Closed serial connection
Traceback (most recent call last):
  File "/usr/local/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flash.py", line 40, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 654, in run_until_complete
[... 10 lines cut ...]
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service banner successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

tube0013, 2025-01-21:
Did you stop ZHA/Z2M?

jkosharek, 2025-01-21:
This is new and not connected to Z2M. 

[image]

jkosharek, 2025-01-21:
I was able to update using this link:
https://github.com/tube0013/tube_gateways/raw/refs/heads/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp/4.4.3/tubesZB-EFR32-MGM24_NCP_7.4.3.gbl

Then I tried this:
https://github.com/tube0013/tube_gateways/raw/refs/heads/main/models/current/tubeszb-efr32-MGM24/firmware/mgm24/ncp/4.4.4/tubeszb-mgm24-hw-max_ncp-uart-hw_7.4.4.0.gbl

And the 4.4.4 update works.

Is there a way to read the version of the device to verify?

q-bhatti, 2025-03-14:
Hi

I am getting the following error: 

 Add-on: TubesZB Silicon Labs FW Flasher
 TubesZB Silicon Labs firmware flasher add-on
-----------------------------------------------------------
 Add-on version: 0.3.4.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 14.2  (amd64 / qemux86-64)
 Home Assistant Core: 2025.3.3
 Home Assistant Supervisor: 2025.03.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
s6-rc: info: service banner successfully started
s6-rc: info: service universal-silabs-flasher: starting
[17:24:11] INFO: Starting universal-silabs-flasher with socket://192.168.0.52:6638 (bootloader baudrate 115200)
2025-03-14 17:24:11.911 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flash INFO Extracted GBL metadata: NabuCasaMetadata(metadata_version=1, sdk_version='4.4.4', ezsp_version='7.4.4.0', ot_rcp_version=None, cpc_version=None, fw_type=<FirmwareImageType.ZIGBEE_NCP: 'zigbee_ncp'>, fw_variant=None, baudrate=115200)
2025-03-14 17:24:11.911 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.GECKO_BOOTLOADER at 115200 baud
2025-03-14 17:24:13.926 d0f5a498-tzb-silabs-flasher universal_silabs_flasher.flasher INFO Probing ApplicationType.EZSP at 115200 baud
Traceback (most recent call last):
  File "/usr/local/bin/universal-silabs-flasher", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flash.py", line 37, in inner
    return asyncio.run(f(*args, **kwargs))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 654, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flash.py", line 353, in flash
    await flasher.probe_app_type()
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/flasher.py", line 149, in probe_ezsp
    async with self._connect_ezsp(baudrate) as ezsp:
  File "/usr/local/lib/python3.11/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/universal_silabs_flasher/emberznet.py", line 27, in connect_ezsp
    await ezsp.connect(use_thread=False)
  File "/usr/local/lib/python3.11/site-packages/bellows/ezsp/__init__.py", line 138, in connect
    await self.startup_reset()
  File "/usr/local/lib/python3.11/site-packages/bellows/ezsp/__init__.py", line 119, in startup_reset
    await self._gw.wait_for_startup_reset()
  File "/usr/local/lib/python3.11/site-packages/bellows/uart.py", line 63, in wait_for_startup_reset
    await self._startup_reset_future
ConnectionResetError: Remote server closed connection
[17:24:13] INFO: universal-silabs-flasher-up script exited with code 1
s6-rc: warning: unable to start service universal-silabs-flasher: command exited 1
s6-rc: info: service banner: stopping
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
s6-rc: info: service banner successfully stopped
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

tube0013, 2025-03-16:
I'll need to look into this.