tube0013/tube_gateways #228: POE gateway with just an USB?

jankornblum, 2025-04-18:
Related to your Z-Wave PoE Kit, have you ever thought about releasing a similar gateway (based on Olimex POE with ser2net implementation), which has just an USB port instead of the fixed Z-Wave radio? That would allow users to use different controllers and sticks.

tube0013, 2025-04-18:
I am not sure how feasible this would be. All zwave sticks are serial based and just use a chip to go to usb. It may work with an esp32-s3 as espressof has a dev kit with usb-otg. I can definitely keep an open mind to this though.

Thanks for the suggestion.

tube0013, 2025-04-18:
Well you got my mind racing. And proud of concept works (using zigbee module for testing) https://github.com/user-attachments/assets/ebfbbda0-13b5-4bdb-96bc-c058105fd474

jankornblum, 2025-04-19:
That’s cool! I’ve written to Olimex if there are any UEXT to USB adapters or any different options (don’t know enough myself about this hardware stuff, but I really like their POE boards).

tube0013, 2025-04-19:
Unfortunately the current Olimex boards with UEXT do not have an esp-s3 which is needed for this. Also the uext header does not expose the GPIO pins needed for the usb-otg connection (19 and 20)

jankornblum, 2025-04-21:
Too bad… So i guess the other PoE boards like Wesp32 or LilyGo are not an option either?

tube0013, 2025-04-21:
you can do it with the lilygo t-eth lite: https://mastodon.social/@tubeszb/114364548320172114

I have an esp32-s3 based board in early development. while it doesn't currently have a USB A port on it, I will see if using an OTG cable with the usb-c port works - I'm not sure the voltage will go out the way it is currently set up.