tube0013/tube_gateways #231: Cannot connect to zigbee radio with TubesZB Dual Radio Kit MGM24?

ClintElliotMalcolm, 2025-06-06:
I just received my dual radio ezsp and I've been trying to connect to it but I haven't been able to the zigbee radio in any way.

I can see the web ui just fine. However, when I try connecting to it from zigbee2mqtt I get failures with this log.

```
[18:50:48] INFO: Preparing to start...
[18:50:48] INFO: Socat not enabled
[18:50:48] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-06-05 18:50:49] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-06-04.19-37-10'
[2025-06-05 18:50:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-06-05 18:50:49] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-06-05 18:50:49] info: 	z2m: Starting Zigbee2MQTT version 2.4.0 (commit #unknown)
[2025-06-05 18:50:49] info: 	z2m: Starting zigbee-herdsman (4.1.0)
[2025-06-05 18:50:49] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[24],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"tcp://192.168.1.250:6638\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-06-05 18:50:49] info: 	zh:ember: Using default stack config.
[2025-06-05 18:50:49] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[24]},"serialPort":{"path":"tcp://IP:6638","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-06-05 18:50:49] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-06-05 18:50:49] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-06-05 18:50:49] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-06-05 18:50:49] debug: 	zh:ember:uart:ash: Opening TCP socket with IP:6638
[2025-06-05 18:50:49] debug: 	zh:ember:uart:ash: Socket connected
[2025-06-05 18:50:49] info: 	zh:ember:uart:ash: Socket ready
[2025-06-05 18:50:49] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-06-05 18:50:49] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2025-06-05 18:50:49] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2025-06-05 18:50:50] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2025-06-05 18:50:51] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=0)
[2025-06-05 18:50:51] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-06-05 18:50:51] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0](ackRx=0)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=0](frmRx=0) Added to rxQueue
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=1)
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: <=== [FRAME: ID=0:"VERSION" Seq=0 Len=7]
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=1 Len=6]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=1](ackRx=1)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=1](frmRx=1) Added to rxQueue
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=2)
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: <=== [FRAME: ID=0:"VERSION" Seq=1 Len=9]
[2025-06-05 18:50:51] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-06-05 18:50:51] debug: 	zh:ember: Adapter info: EZSPVersion=13 StackType=2 StackVersion=29760
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: ===> [FRAME: ID=170:"GET_VALUE" Seq=2 Len=6]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=2 frmRx=2](ackRx=2)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=2 frmTx=3)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2](frmRx=2) Added to rxQueue
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=3)
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: <=== [FRAME: ID=170:"GET_VALUE" Seq=2 Len=14]
[2025-06-05 18:50:51] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-06-05 18:50:51] debug: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=3 Len=8]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=3](ackRx=3)
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-06-05 18:50:51] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=3 frmTx=4)
[... 235 lines cut ...]
    return future.result()
  File "/home/clint/.local/lib/python3.9/site-packages/universal_silabs_flasher/flash.py", line 269, in write_ieee
    await ctx.obj["flasher"].write_emberznet_eui64(ieee, force=force)
  File "/home/clint/.local/lib/python3.9/site-packages/universal_silabs_flasher/flasher.py", line 344, in write_emberznet_eui64
    await self.probe_app_type(
  File "/home/clint/.local/lib/python3.9/site-packages/universal_silabs_flasher/flasher.py", line 229, in probe_app_type
    result = await probe_funcs[probe_method](baudrate=baudrate)
  File "/home/clint/.local/lib/python3.9/site-packages/universal_silabs_flasher/flasher.py", line 149, in probe_ezsp
    async with self._connect_ezsp(baudrate) as ezsp:
  File "/usr/lib/python3.9/contextlib.py", line 175, in __aenter__
    return await self.gen.__anext__()
  File "/home/clint/.local/lib/python3.9/site-packages/universal_silabs_flasher/emberznet.py", line 27, in connect_ezsp
    await ezsp.connect(use_thread=False)
  File "/home/clint/.local/lib/python3.9/site-packages/bellows/ezsp/__init__.py", line 138, in connect
    await self.startup_reset()
  File "/home/clint/.local/lib/python3.9/site-packages/bellows/ezsp/__init__.py", line 119, in startup_reset
    await self._gw.wait_for_startup_reset()
  File "/home/clint/.local/lib/python3.9/site-packages/bellows/uart.py", line 63, in wait_for_startup_reset
    await self._startup_reset_future
ConnectionResetError: Remote server closed connection
```

I reflashed it with the "TubesZB Dual Radio Kit MGM24" firmware but no change.

Possibly related: the device is discovered by home assistant but is not discovered by esphome. This was unexpected and seems like a bug too.

ClintElliotMalcolm, 2025-06-06:
Sorry I was unclear I had already turned off z2m when I tried the flasher

tube0013, 2025-06-06:
Try while not having the web front end open. #230 seemed to be running into the same issue. I need to investigate if something changed as in testing I could reliably have 3 connections open. But it seems like it is 2 for some reason now. I may need to pin to an older esphome version if something there changed.