tube0013/tube_gateways #234: Stability Issues with TubesZB Dual Radio Kit with MGM24 and ZAC93

rmpratt1, 2025-07-06:
I recently got a TubesZB Dual Radio KIT with MGM24 and ZAC93. The device keeps crashing and I have to power cycle the POE port to get it back up and running. I’m not sure how long it’s staying running for but it’s been less than 24 hours per time.

tube0013, 2025-07-06:
Hi, what kind of network router are you using?

rmpratt1, 2025-07-06:
Entire network is Unifi equipment, seen it happen on both my core POE+ switch (USW-24-POE) and a remote POE+ switch (USW-FLEX-MINI).

Also realized I should give firmware versions:
MG24 - 7.4.4.0
ZAC93 - FW 2.2 SDK 7.23.2

tube0013, 2025-07-06:
I have a similar set up. is it on the same or different VLAN as HA/ZWaveJS? 

you can try using an ESPHome build with a static IP, even when the device has a reserved IP on the network other has found that this can help with disconnects. I have a GitHub Repo that can build the firmware for you here: https://github.com/tube0013/TubesZB-ESPHome-Builder Just use the issue template to enter the details.

rmpratt1, 2025-07-24:
I'm going to mark this issue closed.

It does seem like it was a flaky ESP32-POE. I tried updating the one it came with a few times and now seems to be having even more issues. The replacement board is working as expected.

closed 2025-07-24 by rmpratt1