tube0013/tube_gateways #237: Process to upgrade Zooz ZAC93 firmware

onedr0p, 2025-07-31:
I am trying to follow the docs in https://tube0013.github.io/tube_gateways/zwave-kits/#zooz-zac93-module but I do not see a firmware upgrade  link.

My current versions:

FW: v1.0
SDK: v7.18.1

[image]

and under firmware upgarde, there's no option to upload a firmware.

[image]

onedr0p, 2025-07-31:
Related: https://github.com/zwave-js/firmware-updates/issues/176

tube0013, 2025-07-31:
https://www.support.getzooz.com/kb/article/1158-zooz-ota-firmware-files/

has a download link for the latest ZAC93 firmware.

onedr0p, 2025-07-31:
Not what I asked but I will continue this conversation on the linked issue.